alejoseb/Modbus-STM32-HAL-FreeRTOS #2: Problems with higher Baudrates on STM32F303K8

aq-daniel, 2020-10-30:
Hello alejoseb,

thank you for your great work of converting the modbuslibrary to STM32 HAL and FreeRTOS. I managed to copy your example for the F429 Modbus Slave to my STM32F303K8 Microcontroller. I'm using the serial port on uart2, which is routed through the usb on my development board to communicate with the F303 via QModbus on my PC. After changing the Priority of the global UART Interrupt to 5, everything seems to work fine at a Baudrate of 38400 kbit/s. But if I increase the baudrate to 115,2 kbit/s or even 57,6 kbit/s it stops working. Any ideas why this could be the case?
My Investigations of the problem lead to the ModbusSlaveTask(), where the RxBuffer is recieved (with getRxBuffer()-method), it gives me only 6 Bytes@57,6kbit/s and 2 Bytes@115,2kbit/s as return value and as these are invalid Framesizes the Message is not parsed further and a BAD_SIZE-Error is thrown. 
If I understand your code correctly - pls correct me if i'm wrong - you are receiving every Byte from the UART through a seperate Interrupt and then sending it off to the ModbusQueue (at modH->QueueModbusHandle). The SlaveTask() is only notified if the xTimerT35-Callback is called, right? Why would it execute if it has not recieved all bytes yet?

I'm running my MCU at 64 MHz, which should be plenty for 115,2 kbit/s in my opinion. 

I would be grateful for any tips to get a better understanding of your Modbus stack and my problem itself. If you need more information, feel free to ask, please.

Daniel

alejoseb, 2020-10-30:
Hi,
I pushed a new example for the STM32F303RE working at 115200 kbps without any issue. I suppose you are using the ST-Link and the provided USB-to-serial converter, I used the same configuration with Qmodbus.

You are right the global interrupt needs to be changed to a lower preemption priority (higher number) than the FreeRTOS kernel, i.e. 5.  I included that detail in the Readme.

xTimerT35-Callback is called whenever no more consecutive bytes are received (timeout), this means that either the whole frame was received or only a fragment of it was received. In the former case, the whole frame stored in the queue will be processed by the Modbus stack, in the later case the frame will be discarded, which is your case.  

I am not sure what could be the reason, if you are using more interrupts, I recommend to keep the USART at a higher preemption priority than other interrupts but the FreeRTOS.

aq-daniel, 2020-11-02:
Thanks for your quick reply and the new example for F303. I did some more testing and found a problem in my clock tree configuration, which lead to an inaccurate clock frequency and my issue with the UART-Callback at higher baudrates.
I'll close this issue. Thanks again for your great work :)

closed 2020-11-02 by aq-daniel