alejoseb/Modbus-STM32-HAL-FreeRTOS #4: Corrupted answer from slave

djuseeq, 2021-01-03:
Hi, first of all, thanks for your work to porting that arduino library what i'm using in a couple of devices. I have started to learn programming STM32 devices and FREE_RTOS and exactly this library is that what i need. So i will try to explain what issue i faced. 
The master device is an STM32F103RB MCU with an a [SP485E](https://www.mouser.se/datasheet/2/146/sp481e_sp485e-1816374.pdf) transceiver attached to USART3 (Interrupt enabled, priority 5, 9600bps 8N1). The slave is an a Arduino ProMini (M328) with MAX485 transceiver  on Serial. I have tried with modified example of ModbusF103 (Just master task) and i faced many dropped incoming package due to CRC ERROR (Average ~40% error rate). If i use more tasks (2 extra task to blink LEDs with different interval using osDelay functions in it) then the error rate jumps up to >90%. I found that when the error occurs then the first byte from the answer stream is lost somewhere (Slave ID) and this causes CRC problem (the whole message is came to MCU i have checked with Logic analyzer). 
[image]
[image]

djuseeq, 2021-01-03:
Some update. It's looks like the answer has arrived too quickly and the RX interrupt is missed the first byte. If i insert some delay on arduino side in the sendTxBuffer() then the STM doesn't miss a stream. 
With 5ms delay: one task 1.25% error rate, multiple tasks 86.59%
With 100ms delay: one task 0% . multiple tasks 0% error rate, what is really good number, but this kind of delays is not needed when other arduino is the master. Some timing issue can be on the master side or something with RTOS? Tick rate is default 1000 Hz.
Arduino side, code snippet:
```C
void Modbus::sendTxBuffer()
{
    // append CRC to message
	delay(100); //<---ADDED
    uint16_t u16crc = calcCRC( u8BufferSize );
    au8Buffer[ u8BufferSize ] = u16crc >> 8;
    u8BufferSize++;
    au8Buffer[ u8BufferSize ] = u16crc & 0x00ff;
    u8BufferSize++;

    if (u8txenpin > 1)
    {
        // set RS485 transceiver to transmit mode
        digitalWrite( u8txenpin, HIGH );
    }

    // transfer buffer to serial line
    port->write( au8Buffer, u8BufferSize );

    if (u8txenpin > 1)
    {
```

alejoseb, 2021-01-03:
I am not sure what could be the issue. Also, I have not tested my port with rs485 transceivers controlling the DE/RE signals. The  routine is implemented but not verified (that is in the TODOs section).  It is possible that DE/RE is not controlled appropriately and that  interferes with the reception of bytes. 

Also, some users reported a high error rate because of a wrong system clock configuration. You should verify that your system clock is high enough to manage all the tasks and it is using a stable clock source, for example a crystal.  

Which development board are you using?

djuseeq, 2021-01-03:
Hi, i have [this](https://www.aliexpress.com/item/32809724194.html) board with integrated RS485 interface. On this board i don't have to control the DE/RE pins, i use like normal RS232, a small circuitry handle these lines automatically(if TX signal is active then a transistor put the transceiver in transmit mode, otherwise is in receive mode). I have used with max 72MHz when i faced this issue, but for testing purpose tried with one task 115200bps and 16MHz system clock and with the delay on slave side (10ms is enough) works without error. With 8MHz is unusable. Maybe then when I see how the code works, i will change the CMSIS_RTOS functions for native FREE_RTOS functions in the library and the i will check with SEGGER what is going on during running the code.

alejoseb, 2021-01-04:
I reproduced your experiment and I found the same issue with a high rate of errors.  I identified the root problem, I found an omission in the notification from the TX complete callback. Please pull my latest commit.
I also have a version ported to the native FreeRTOS functions, but I will publish it later keep tuned to this repository.
I did not publish that version to maintain portability among HAL supported devices.

djuseeq, 2021-01-04:
Sir, you are awesome. The issue is solved. Now it work like a charm, without using the ugly delay on the slave device. :) 
I have added your repo to watch list ;)

closed 2021-01-04 by djuseeq