alejoseb/Modbus-STM32-HAL-FreeRTOS #6: Slave Does not allow for memory space larger then 256

ZARichardWright, 2021-01-10:
First off Great Library so far.

I would like to map more the 256 bytes to a modbus slave. I know this is the limit of any one transaction but still would like to have the option to map more. 
From what i can see the limit is first and for most the fact that u8regsize member of the modbusHandler_t is a uint8_t. 
I have sort got it working by just increasing this to a 16_t as well as all other locations where the address is cast to a 8_t. My question is am i missing another obvious method of allowing for a larger address space. as i would obviously preferer to try avoid deviating form current master.

Cheers

alejoseb, 2021-01-11:
You are right about the  transaction size and the memory map, the uint8_t was used in  the original  Arduino library and I did not change that. 
I think your approach to extend the memory map is correct, there is no other obvious method.

alejoseb, 2021-01-22:
Solved in commit  b2e6aa96

closed 2021-01-22 by alejoseb