alejoseb/Modbus-STM32-HAL-FreeRTOS #8: Communication lost with Honeywell Comfortpoint DDC panel

suyog44, 2021-02-27:
First of all I would like to thanks for your great work. Really appreciate. I have tried some experiment with Honeywell PLC (Comfortpoint DDC panel) as well as with Modscan32 Application. 

When I connected Slave ID 1 And Slave ID2 in Daisy chain topology it works great with Modscan32. Everything run flawlessly.

But When same thing I have tried with Honeywell PLC, there is communication loss after 2 hrs and again reconnected in 30mins. Main problem is only one device os responding at a time. 

Do you have any idea about this problem.

alejoseb, 2021-03-01:
Due to the few details that you shared  and your specific setup it is hard to determine a possible cause. 

You should first test each slave separately and see if the issue is persistent. 
You can connect a debugger to the slave that is failing and check the error rate of packets, the modbus handler provides some counters to keep track of errors. Also verify that your slave is not halting or crashing by itself when it is not responding. 

If you can reproduce the issue with development boards, you can share  the details and I will take a deep view. 
Meanwhile I hope you can solve your issue.

suyog44, 2021-03-05:
Honeywell PLC found some internal issues with its application. there was too much CRC error found with PLC itself. might be that was the issue as device is not communicating with PLC. We have tested our code up to 4 to 5 hours on modscan32 there was no single packet drop on it.

alejoseb, 2021-03-08:
Ok, 
I cannot help with that then. If you have more code running besides the modbus library  that could be the reason. 
If you identify something more specific  and a method to reproduce the problem feel free to reopen the issue and I will take a look for issues  related to  the modbus library.