alejoseb/Modbus-STM32-HAL-FreeRTOS #18: RS485 dose not work for the second time

houchaoo, 2021-05-27:
the first time is ok ,the data is correct ,but another message does not resoponse

alejoseb, 2021-05-27:
Hi,
I need more details about your issue, setup, hardware and software configuration. Otherwise, I cannot help you.
Thanks

houchaoo, 2021-05-27:
Hi thanks for reply,my mcu is stm32f407vgt6, the 485 chip is ADM2587EBRWZ,the setting is like this below.

	ModbusH.uModbusType = MB_SLAVE;
	ModbusH.port = &huart3; // This is the UART port connected to STLINK in the NUCLEO F429
	ModbusH.u8id = 1; //slave ID
	ModbusH.u16timeOut = 1000;
	ModbusH.EN_Port = RS485_Ctrl_GPIO_Port;
	ModbusH.EN_Pin=RS485_Ctrl_Pin;
	//ModbusH2.EN_Port = LD2_GPIO_Port; // RS485 Enable
	//ModbusH2.EN_Pin = LD2_Pin; // RS485 Enable
	 ModbusH.xTypeHW = USART_HW;

	ModbusH.u16regs = ModbusDATA;
	ModbusH.u16regsize = sizeof(ModbusDATA) / sizeof(ModbusDATA[0]);
	//Initialize Modbus library
	ModbusInit(&ModbusH);
	//Start capturing traffic on serial Port
	ModbusStart(&ModbusH);

[image]

the first time when i send the read message ,i can get a reply form my mcu,however when i try to read again ,the mcu reply nothing,and another time i try to read ,the mcu works ok, and so on ....

houchaoo, 2021-05-27:
[image]
[image]
[image]
[image]
[image]
[image]

houchaoo, 2021-05-27:
RS232 works fine..

houchaoo, 2021-05-27:
I find that when i switch the RS485 ctrl pin ,I got a message "0x00" may be the 485 chip is not good . thanks a lot

alejoseb, 2021-05-27:
Yes probably that is the problem. 
I have tested with transceivers MAX485 without any issue, I verified the control signal with an oscilloscope. 
I have also used the MAX1438, which does not require the control signal.

houchaoo, 2021-05-29:
when I change the T35 from 5 to 6,the 485 works fine...thanks a lot ,Your project is wonderful

alejoseb, 2021-05-30:
Glad to know this library is useful for your project, I will close this issue then.

closed 2021-05-30 by alejoseb