alejoseb/Modbus-STM32-HAL-FreeRTOS #20: Fail to access a register address above 255

egoltzman, 2021-05-29:
Hi,
I have 2600 registers (40000-42600)
I fail to read above 4256
I see that in the code you put the 16bit address into an 8bit variable, is this OK?
 
```
int8_t process_FC3(modbusHandler_t *modH)
{

    uint8_t u8StartAdd = word( modH->u8Buffer[ ADD_HI ], modH->u8Buffer[ ADD_LO ] );

```

alejoseb, 2021-05-29:
You are right, this is part of the original Arduino library and its limitations. I will update the library soon.

alejoseb, 2021-05-30:
Hi @egoltzman I double checked Modbus documentation and the maximum size of a MODBUS RTU frame is 256 bytes. Therefore, to fulfill your requirement you can issue two Modbus queries to cover the range of registers that you need, or change the library by yourself. However, this last option will be out of specification, therefore I will keep the library in this repository as is.

egoltzman, 2021-05-30:
Hello @alejoseb, thank you for getting back to me on this.
I don't think the problem I'm pointing refer to the frame size.
I refer to the register address that is built out off two bytes, the ADD_HI and ADD_LO and in word() you build a 16bit address value out of those two bytes, but the result is stored in u8StartAdd and thus the address value is truncated to 8bit and the lib does not let the user access registers above 256.
No?

alejoseb, 2021-05-30:
@egoltzman You are right, I will check what are the implications of using the right casting in the library, since the frame size must be keep regardless of the addressing scheme.

alejoseb, 2021-06-07:
Hi @egoltzman  I updated the library, now it is possible to read/write registers using a 16-bit address. Besides, I added the validation to maintain the 256-byte frame limit regardless of the register's address. You can close the issue after you validate it, or I will close it by the following week.

closed 2021-06-14 by alejoseb