alejoseb/Modbus-STM32-HAL-FreeRTOS #22: TCP Client library usage example

Frenelius, 2021-06-29:
Hello,
I am trying to use your library in a project but I do not really know hoy to use it in a thread.

Everything compiles fine, but it will not connect to the MTCP server on my LAN (which works with other devices).

Could you explain how to assign the server IP with xIpAddress?

Do you have an example for the library usage? I cannot see how to use it in a RTOS thread.

Thank you!

alejoseb, 2021-06-30:
Hi,
I provide the complete source code of 2 examples (including the CubeMX configuration file) for the Núcleo boards F429 and H743. These examples include the usage of the library in a FreeRTOS task. 

You can check this file of the F429 for your particular question: Examples/ModbusF429TCP/Core/Src/freertos.c

To assign the IP address, you can either launch the CubeMX tool of CubeIDE and change the TCP Address there, or manually change the IP in the CubeMX generated files.

Frenelius, 2021-06-30:
Thank you so much!

I was blind enough to not look into freertos.c haha.

Now I got it working in my F407VET.

The only thing that happens now is that the software gets stuck into prvCheckTasksWaitingTermination() but I think it is not related to your Modbus library.

alejoseb, 2021-06-30:
No problem, close the issue when you solve your issue and do not forget to star  :star: the library project  :smiley:

Frenelius, 2021-07-01:
I forgot to ask, I have a compiler error in modbus.c, lines 479, 892 and 1479 for netconn_set_sendtimeout.

The error is undefined reference to 'netconn_set_sendtimeout'

If I comment the 3 lines, it builds OK.

Do you know if I have to copy some other files?
Your example for the F429 builds fine including the three lines.

I have started a completely new project with just your library just in case it was another part of the code, but it happens the same.

Thank you,

alejoseb, 2021-07-01:
You have to enable the timeout APIs of LWIP for RX and TX in the option file.

For example:
https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/fc3faba06efafbe6e896997f64505ea2b9550f90/Examples/ModbusF429TCP/Middlewares/Third_Party/LwIP/src/include/lwip/opt.h#L1997

Frenelius, 2021-07-02:
Hi Alejoseb, 

It has worked and now compiles OK.

But it keeps getting stuck randomly when it goes into "xQueueSemaphoreTake" (queue.c line 1418), and assigning
configASSERT( pxQueue->uxItemSize == 0 );

Could it be related to the line in 154 in FreeRTOSConfig.h:  ??
#define configASSERT( x ) if ((x) == 0) {taskDISABLE_INTERRUPTS(); for( ;; );}

Thank you,

alejoseb, 2021-07-02:
Hi,
I have seen this assert  "configASSERT( pxQueue->uxItemSize == 0 );" failing on errors related to running out of heap memory and buffer overflows that corrupt the semaphores, you should double check that you have enough memory assigned to each task, try incrementing the heap of your tasks.

PS1: Probably, I cannot verify it.
PS2: Try keeping  MX_LWIP_Init() where CubeMX put it, instead move 
 ModbusInit(&ModbusH); and   ModbusStart(&ModbusH); after the call to MX_LWIP_Init();

Frenelius, 2021-07-05:
Now the MX_LWIP_Init(); location works well, assigning a higher priority to de StartDefaultTask.

The only thing that keeps giving error is the "configASSERT( pxQueue->uxItemSize == 0 );", but I think it is not related to your code, so I am closing this issue now.

Thank you!

closed 2021-07-05 by Frenelius