alejoseb/Modbus-STM32-HAL-FreeRTOS #23: H743 master issue

emreinanici, 2021-06-30:
I can't set my device as master.  I used slave example.

        ModbusH.uModbusType = MB_MASTER;
        ModbusH.port =  &huart3; // This is the UART port connected to STLINK in the NUCLEO F429
        ModbusH.u8id = 0; //slave ID
        ModbusH.u16timeOut = 1000;
        ModbusH.EN_Port = NULL; // No RS485

NOT COMMUNICATING WITH THE PC !  Can you provide detailed information for Master device settings ?

alejoseb, 2021-06-30:
Hi, 
I provide TCP and RTU project examples for the Nucleo H743. You can check there the configuration in the main.c file. 

Also, an example of master configuration can be found in the following file:

Examples/ModbusF103/Core/Src/main.c

Remember to select the type of hardware for USART, USB CDC, or TCP.

emreinanici, 2021-07-04:
Hi alejoseb .
I already checked the given examples. I also changed the freertos.c file accordingly F103 file. But it didn't work

alejoseb, 2021-07-05:
You should provide more information about your setup, so I can help if possible. The only thing that I know is that you are implementing a master, which is not working.

closed 2021-07-06 by alejoseb