alejoseb/Modbus-STM32-HAL-FreeRTOS #26: problems with master mode uart hw

micheledarold, 2021-08-06:
Using master rtu mode with uart I have problem because the rx buffer contains both tx bytes and rx bytes. The modH->u8BufferSize is the sum of the two parts and so the reception doesn't work.

with old version (without ring buffer) everything work but I need a different HAL_UART_TxCpltCallback

void HAL_UART_TxCpltCallback(UART_HandleTypeDef *huart)
{
  /* Modbus RTU callback BEGIN */
  //BaseType_t xHigherPriorityTaskWoken = pdFALSE;
  int i;
  for (i = 0; i < numberHandlers; i++) {
    if (mHandlers[i]->port == huart) {
      //xTaskNotifyFromISR(mHandlers[i]->myTaskModbusAHandle, 0, eNoAction, &xHigherPriorityTaskWoken);
      break;
    }
  }
  //portYIELD_FROM_ISR( xHigherPriorityTaskWoken )

the new version with RingBuffer is not working, (the same with original HAL_UART_TxCpltCallback or the modified one posted here)

I'm using STM32F767 microcontroller. Baudrate is 57600 8n1.

alejoseb, 2021-08-06:
Hi,
"the rx buffer contains both tx bytes and rx bytes. The modH->u8BufferSize is the sum of the two parts" that statement is not true. 

The library uses the ring buffer exclusively for  rx. The u8Buffer on the other hand, is used as a working buffer to copy the rx frame and to prepare the tx frame, both operations occur at different times and they are mutually exclusive. Therefore, your issue may not be related to what you mentioned. 
Eventually, your issue is related to the way you are using the master handler and the notifications, double check the examples that I provided. If you have a piece of code to reproduce your issue I will take a look, otherwise I cannot provide more help since the library is validated in multiple boards for the interrupt and DMA modes.

alejoseb, 2021-08-16:
Since there is no activity here, I am closing this issue. Feel free to reopen this issue if you have a piece of code that I  can use to reproduce your issue.

micheledarold, 2021-08-24:
Hi @alejoseb 
we was on vacation and I haven't see your post.
we don't have modification in UARTCallback.c, the file is the same of the git one.
I think the point is related to hardware. We have a RS485 communication line so the hardware setup cause the cpu to receive its tx in rx pin.

alejoseb, 2021-08-25:
Hi,
I have tested with multiple slaves and a single master using MAX485 transceivers. Your issue is similar to issue https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/issues/21 you should read the whole conversation there.

micheledarold, 2021-08-25:
Hi @alejoseb 

I think the problem is related to our hw configuration. we have tied to GND the RE pin so the receiver is always enabled
[image]

in this condition I have the received enabled also during our transmission and in the ring buffer I have the reply of slave followed by the next answer from master to slave, in order to resolve the issue I disable the rx irq during the trasmission with the attached patch with our modification.
[ModbusLibPatchForRXdisable.zip](https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/files/7045552/ModbusLibPatchForRXdisable.zip)

We are now in test with this modification and everything seem to work fine.

I don't know if you are in the same condition of us with the RE pin.

alejoseb, 2021-08-25:
Hi, 
My connection is a bit different, RE and DE are tied together. Disabling the RX irq during TX is the right solution, but not implemented since latest RS485 transceivers don’t even need to control RE and DE. You can send me a pull request with your patch so I can incorporate your changes.

alejoseb, 2021-08-30:
Hi, 
I modified the library to disable the RX IRQ during TX. I added this modification for the DMA mode as well. 
You can check the latest commit.

micheledarold, 2021-08-31:
Hi @alejoseb your new implementation doesn't work on my setup. The received size is always 0.

alejoseb, 2021-08-31:
I pushed a new version that I tested using your same circuit connection. It works without issues with the master. The slave also works but it triggers a spurious interrupt that I need to investigate a bit more. The slave handles the spurious interrupt but ideally it should not appear. In any case, connecting DE and RE together solve the issues for master and slave using the current or previous versions of the library.

closed 2021-09-09 by alejoseb

micheledarold, 2021-09-15:
Hi @alejoseb the latest version work well but sometimes the connection stop working, the packet is sent and the slave respond (I check the communication with a oscilloscope and the packet are good.
I think the problem could be related to error management, when the serial got a error (maybe overrun) could be better to reset RingBuffer, discard the packet and restart?
I mean to add HAL_UART_ErrorCallback and inside reset and restart