alejoseb/Modbus-STM32-HAL-FreeRTOS #30: Generated failed in touchgfx

KomKGT, 2021-09-16:
Hi @alejoseb . I use STM32F4-discovery and first step I use touchgfx to generated code but the touchgfx is fail to generated and show that " Generate Code
    Generate
        Wrote config/gcc/app.mk
        Done
    Generate Assets
        make -f simulator/gcc/Makefile assets -j8
        process_begin: CreateProcess(NULL, ../Middlewares/ST/touchgfx/framework/tools/imageconvert/build/win/imageconvert.out -r assets/images -w generated/images, ...) failed.
        make (e=2): The system cannot find the file specified.
        
        make[1]: *** [images] Error 2
        generated/simulator/gcc/Makefile:215: recipe for target 'images' failed
        simulator/gcc/Makefile:32: recipe for target 'assets' failed
        make: *** [assets] Error 2
        Failed
    Failed "
How to fix it ? Please give me some advice I am a newcomer to stm32 (sorry about my english)

alejoseb, 2021-09-16:
Sorry, I cannot help with that, your  issue is beyond the scope of this library. I cannot provide support with touchgfx since my knowledge of that tool is basic, I would say enough to build the demo that is on the examples, but no more than that.

closed 2021-09-16 by alejoseb

bnina-ayoub, 2024-01-20:
Hey @alejoseb, It's been a while since your problem, I had the same issue but I found that deleting unsupported files from the assets folder will solve it.