alejoseb/Modbus-STM32-HAL-FreeRTOS #32: modbus TCP Slave tool test is showing errors

aboud96, 2021-09-29:
Hi,

Thank you @alejoseb for the work you did under this library and I found it very useful.
I'm using modbus tcp slave and master for my stm32h7,  when I tested the slave mode the Modbus Poll tool is showing : Read Error and Write Error after every poll ,so I tested the master and slave at the same time in order to store the incrementing value at the same register ModbusDATA but the errors are blocking the incrementation in modbus poll tool, Do you think that is related to the fact that in the slave mode, the slave is disconnecting after every poll?

Looking for your help

aboud96, 2021-09-29:
[image]
[image]

alejoseb, 2021-09-29:
Hi, 
The library effectively closes the TCP connection after every request. Probably that is the reason as you mentioned.

That behavior is not "recommended", but still  valid. If the tool that you are using is not able to manage the disconnection from the slave, then that tool is not modbus compliant. You can read the details in issue #23.

You can use QModbus tool, and check how a Modbus compliant master can deal with TCP disconnection without issues.