alejoseb/Modbus-STM32-HAL-FreeRTOS #35: MB-TCP  (Using ETH-TO-SERIAL module)

mojtabamza, 2021-11-13:
Hi Alejandro,
Thank you for sharing the Library.
I am working on Modbus TCP and I have come across some problems.
The MCU is STM32L151RE and I want to use wiz105SR (ETH-TO-SERIAL) module for making connections between MCU and network. So I used the MODBSUF429TCP example and am trying to modify it for my system.
Considering the system as the slave, how can I modify the library?

alejoseb, 2021-11-15:
The TCP example is meant to work with the LWIP stack, since you are using the wiz105SR module I think that will not work because my implementation relies on  specific LWIP APIs, probably the USART  example is more adequate for your needs.

Besides that, to make the slave work it is necessary to notify the Modbus task when a frame has arrived by any method (USB, TCP, USART).  For example for USART in line https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/680ca775bf3fff4b5dbbb41e3171dd35a743f0ba/MODBUS-LIB/Src/Modbus.c#L673 the modbus slave receives the notification when the library detects the end of frame (i.e, a silence), and in line https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/680ca775bf3fff4b5dbbb41e3171dd35a743f0ba/MODBUS-LIB/Src/Modbus.c#L675 the function "getRxBuffer" get all the bytes that were received. In the case of TCP the function TCPwaitConnData replaces the notification and the getRxBuffer all together. You need to study these points  for your implementation. Also check the usart callbacks.

alejoseb, 2021-11-29:
I will close this issue, feel free to open it again if you need.

closed 2021-11-29 by alejoseb