alejoseb/Modbus-STM32-HAL-FreeRTOS #42: Receiving UART data from a UART port different than the Modbus port causes Modbus to stop working

emptor8, 2021-12-18:
Hi there,

Thanks for this great library! I'm seeing the following problem:

- If I only use the Modbus library on UART port 1, everything works well.

- If I enable UART port 2 to output messages via `HAL_UART_Transmit(&huart2, data, size, timeout)`, Modbus on UART port 1 still works fine. 

- However, once I try to receive input on UART port 2 (e.g. via `HAL_UART_Receive(&huart2, data, size, timeout)`, Modbus on UART port 1 no longer works. 

Do you know what I'm missing here? 

Note: I added conditional checks in UARTCallback.c to only run the code logic in the callbacks for the &huart1 instance. If the &huart2 instance is detected, the callbacks do nothing.

alejoseb, 2021-12-21:
Hi, 
If you have a complete example to reproduce your issue I can take a look. There could be multiple reasons for this issue. For example, If the task where you are reading data has higher priority than the modbus task

emptor8, 2021-12-22:
Here's a complete example that reproduces the issue. 

I set up the modbus library on one interface, `huart4`. 
I then setup another task (`debugTask`) that reads from and outputs on a different UART interface, `huart5`.

[modbus_test_12.22.21_v1.zip](https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/files/7760253/modbus_test_12.22.21_v1.zip)

emptor8, 2021-12-22:
Attached is a complete example that demonstrates the issue I'm having.

- debugTask1 waits for input from the keyboard via `HAL_UART_Receive()` on `huart5`
- debugTask2 toggles an LED on the NUCLEO board
- Modbus slave task is running and using `huart4`

Things work fine if I comment out `HAL_UART_Receive()` -- I can still transmit UART data on `huart5` and run the Modbus on `huart4` at the same time.

However, with `HAL_UART_Receive()` on `huart5`, it prevents Modbus from running on `huart4`. The LED is still being toggled in debugTask2 though, so it seems like the CPU isn't being held up by `HAL_UART_Receive()`? I've also made debugTask1 and debugTask2 lower priority than the Modbus slave task but this does not help.

[modbus_test_12.22.21_v1.zip](https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/files/7763038/modbus_test_12.22.21_v1.zip)

emptor8, 2022-01-13:
Hey @alejoseb, did you get a chance to reproduce this?

alejoseb, 2022-01-15:
I cannot reproduce your issue since I don't have your board, and you did not include the modified version of the modbus library in the zip it is only a link. I quickly checked the code though. I saw that you use portMAX_DELAY for waiting for a character, that is wrong. portMAX_DELAY is a freertos macro, it is not meant to work with the HAL library, I don't know what could be the effect of using that macro. Besides, you should use interrupt mode to receive the character no the blocking mode.  The blocking mode does not yield. Another option could be to use the blocking mode but with a few ms timeout and check the returned value of the HAL function. HAL_OK should mean that you received a character.

I am closing this  issue since  it is not related to the library.

closed 2022-01-15 by alejoseb