alejoseb/Modbus-STM32-HAL-FreeRTOS #43: 1 Master and Multiple Slave Error is back?

emptor8, 2021-12-20:
Hmm is anyone else seeing the problem noted here? https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/issues/17

This issue seems to have re-appeared on the latest version of this library.

emptor8, 2021-12-21:
An update: Adding a small delay after swapping from transmitter to receiver and vice versa seems to "fix" this issue. Is there a better way to handle this? It seems like maybe the transmitter/receiver are being switched too quickly?

```
    //enable transmitter, disable receiver to avoid echo on RS485 transceivers
    HAL_HalfDuplex_EnableTransmitter(modH->port);
    HAL_GPIO_WritePin(modH->EN_Port, modH->EN_Pin, GPIO_PIN_SET);
    osDelay(10);
```

```
     //return RS485 transceiver to receive mode
    HAL_GPIO_WritePin(modH->EN_Port, modH->EN_Pin, GPIO_PIN_RESET);
    //enable receiver, disable transmitter
    HAL_HalfDuplex_EnableReceiver(modH->port);
    osDelay(10);
```

alejoseb, 2021-12-21:
I don't believe switching to quickly is the reason. It is possible, however, that you are using a very high baudrate. 
If the MCU is not able to serve the ISR on time because of the high baudrate, the USART can overflow. 
If that  is the case, then the  USART will stop working. The library requires a callback to handle this type of error, it is missed right now. 
You can create this callback and check if it is executed, if that is the case, then in the callback you need to clean
the USART state, clean the ring buffer and  start again the USART to receive the modbus data.

emptor8, 2021-12-21:
I'm using a baud rate of 115200. Would that be considered too high?

Where/how would I go about creating this callback?

alejoseb, 2021-12-21:
That baudrate has been tested without issues. 
Check the UARTCallback.c file of the library it already has an error callback but only enabled for DMA mode, you can create  a callback that works also with the satndard mode.

emptor8, 2021-12-22:
Thanks.

Do you know what's the best way to clean the USART state (e.g. which HAL functions)? Is there a difference in what I need to do if I'm using DMA? 

For cleaning the ring buffer, I assume you mean calling `RingClear()`?

emptor8, 2022-01-13:
Hey @alejoseb, sorry to bother you - did you get a chance to look at my question?

saeidshokoufi, 2022-05-07:
Dear 

Thanks for your Library,

 How to set start address of Registers in modebus in this library?

Best Regards

closed 2022-05-11 by alejoseb