alejoseb/Modbus-STM32-HAL-FreeRTOS #44: error in modbus poll "insufficient bytes received"

elimsjxr, 2021-12-29:
Hello, I am a beginner to freertos, and I need help with some questions that I don’t understand.
I ran the code on the B-L475EIOTA2 development board and used this board as a modbus slave station，and send a response using modbus poll, but an error was reported on the modbus poll:**insufficient bytes received**.
It seems that the board has received the data from the master station, but failed to send the response packet. Why? Where does the source code need to be modified?

mpu0, 2021-12-29:
Hi!
I think we need additional informations to answer your question. Please provide the modbus frames (exchanged data) between the devices.

elimsjxr, 2021-12-29:
I think you are talking about this?
[image]

mpu0, 2021-12-29:
Close... I´m talking about the raw data. You can display it with ModbusPoll.. take a look at the second picture on modbustools page under "Quickstart"..

alejoseb, 2021-12-31:
Hi, 
I don't quite understand what are you doing an what is the issue. You should use any of the provided examples as an starting point and build on top of it.

AKanTam, 2023-01-19:
Hello, I had the same problem. 
[image]
this is the modbus message

`
  uint16_t ModbusDATA[100]; 

  ModbusH.uModbusType = MB_SLAVE;
  ModbusH.port =  &huart2;
  ModbusH.u8id = 1;
  ModbusH.u16timeOut = 1000;
  ModbusH.EN_Port = RS485_GPIO_Port;
  ModbusH.EN_Pin = RS485_Pin;
  ModbusH.u16regs = ModbusDATA;
  ModbusH.u16regsize= sizeof(ModbusDATA)/sizeof(ModbusDATA[0]);
  ModbusH.xTypeHW = USART_HW;

 telegram.u8id = 1;
 telegram.u8fct = 3;
 telegram.u16RegAdd = 0x00;
 telegram.u16CoilsNo = 10;
 telegram.u16reg = ModbusDATA;

`

AKanTam, 2023-01-19:
This problem happens with F103VET6. This problem exists regardless of the serial port baud rate is 9600 or 115200

alejoseb, 2023-01-20:
Hi, I am not sure what is the problem, please open a new issue and provide more context and the steps to reproduce your issue.