alejoseb/Modbus-STM32-HAL-FreeRTOS #51: License clarification

hpux735, 2022-04-18:
Hi,

I'm considering using your library in a closed-source project.  I've read the LGPL, and I'm not sure how to use it in a project such as this.  It seems like it's hard to use LGPL libraries on microcontrollers, because we don't have dynamic linking.  Would you consider providing your interpretation of how this library can be used on a statically-linked microcontroller project?  Alternatively, would you consider a similar library that doesn't encumber in the case of static linking?

Thanks!

alejoseb, 2022-04-18:
Hi,
My interpretation is that you can use this library as you please (statically or dynamically linked) even in commercial close-source projects, and you don't need to open source your project. However, if you modify the source code of the library you need to release the code of the modified library, while keeping the rest of the code close-sourced.  

Besides, this is my interpretation, and I am not an expert on licensing and copyrights so don't take my words as legal advice.

hpux735, 2022-04-18:
Thanks Alejandro,

Given that you're the author of the library, your interpretation is the one that matters the most to me :)

closed 2022-04-18 by hpux735