alejoseb/Modbus-STM32-HAL-FreeRTOS #53: Ethernet Communication with Lwip TCP /IP

anilizgu1, 2022-05-11:
Hello,

I've added the Modbus TCP library with the ModbusH743TCP example for my own project. My board is stm32h743-disc0. I am trying to see the firstly ethernet communication between board to pc over Ethernet peripheral. Software gets stuck into prvCheckTasksWaitingTermination() but I think it is not related to your Modbus library. When I ping over the cmd window I couldn't be able to communicate with the board yet. To do that, which parts of code do I need to check especially? When I got the Cubemx generation file after the same configuration (modbus743TCP) I realize that the freertos.c file is different from the example of the ModbusH743TCP project. I thought it can cause a version of cubeide . End of the day I want to make a modbus tcp communication with this board.Could you check below codes from my own project and for any help ı will be appreciated. Thanks in advance. 

Note: I am a beginner at embedded systems.

1.Main.c File 
2.Freertos.c file

*1. Main.c File*
 `/* USER CODE BEGIN Header */
/**
  ******************************************************************************
  * @file           : main.c
  * @brief          : Main program body

  *
  ******************************************************************************
  */
/* USER CODE END Header */
/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "cmsis_os.h"
#include "lwip.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */

/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN PTD */

/* USER CODE END PTD */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */

#ifndef HSEM_ID_0
#define HSEM_ID_0 (0U) /* HW semaphore 0*/
#endif
/* USER CODE END PD */

/* Private macro -------------------------------------------------------------*/
/* USER CODE BEGIN PM */

/* USER CODE END PM */

/* Private variables ---------------------------------------------------------*/

UART_HandleTypeDef huart3;

/* Definitions for defaultTask */
osThreadId_t defaultTaskHandle;
const osThreadAttr_t defaultTask_attributes = {
  .name = "defaultTask",
  .stack_size = 256 * 4,
  .priority = (osPriority_t) osPriorityNormal,
};
/* Definitions for myTaskMaster */
osThreadId_t myTaskMasterHandle;
const osThreadAttr_t myTaskMaster_attributes = {
  .name = "myTaskMaster",
  .stack_size = 128 * 4,
  .priority = (osPriority_t) osPriorityLow,
};
/* USER CODE BEGIN PV */
modbusHandler_t ModbusH;
uint16_t ModbusDATA[16];
/* USER CODE END PV */
[... 494 lines cut ...]

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */

/* USER CODE END PD */

/* Private macro -------------------------------------------------------------*/
/* USER CODE BEGIN PM */

/* USER CODE END PM */

/* Private variables ---------------------------------------------------------*/
/* USER CODE BEGIN Variables */

/* USER CODE END Variables */

/* Private function prototypes -----------------------------------------------*/
/* USER CODE BEGIN FunctionPrototypes */

/* USER CODE END FunctionPrototypes */

/* Private application code --------------------------------------------------*/
/* USER CODE BEGIN Application */

/* USER CODE END Application */

alejoseb, 2022-05-11:
Sorry, I cannot provide support for a particular implementation.
It is expected that the freertos.c is different since that is not part of the library, it is part of an example, or your implementation.
You should first have a TCP connection working  and tested with ping in your board and then try to port the library on top of that.

anilizgu1, 2022-05-12:
Okay, thank you. I will do the TCP connection working first as you suggest.

alejoseb, 2022-05-26:
I think you did not initialized the modbus handler in main.c. 
You can see the initialization in this example: 

https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/32eb8bd122530acb003e28bcbf0fd2324154208f/Examples/ModbusF429/Core/Src/main.c#L97

anilizgu1, 2022-06-03:
Hi, thanks for your time and effort to support.

I am a little bit stuck, according to the ModbusH743TCP example configuration, I've added to my project. When I tried to Master mode with the Slave simulation tool on the pc (PymodSlave), the debug program comes here as below. What could be the reason ? Thanks.

[image]

alejoseb, 2022-06-03:
As you can read there, that might happen if you are running out of heap. Check the configuration of your project, including the heap for each FreeRTOS task that you are creating.

DmitryWorks, 2022-07-17:
Hello, I'm testing your library on NUCLEO-H743ZI2 for Modbus TCP, I checked everything, for example, writing register reading works. I want to go further, for example if 400000 holding register = 1 turn on the built-in LED on the board by analogy with this video https://www.youtube.com/watch?v=-JRI9GmA0dg only for Modbus TCP, but there are problems when I change the configuration of your project in order to use the built-in LED and generate a new code in the MX Cube, everything stops working at the ping level, tell me what could be the matter? I'm using Cube MX 6.3.0 as you created the project. Thank you.

closed 2022-08-03 by alejoseb