alejoseb/Modbus-STM32-HAL-FreeRTOS #57: unknown type errors when trying to use library

krupis, 2022-08-08:
Hello. I have downloaded your modbus library and I am trying to use it. I use STM32F407. I have followed the steps that you have suggested:

1. In CUBEMX, configured UART4 which I want to use for MODBUS
2. Activated global interrupt
3. Enabled DMA RX TX 
[image]
4. Imported the MODBUS-LIB to my project, updated include path in project properties
5. I have renamed ModbusConfigTemplate.h to ModbusConfig.h

When I try to build the project, I get 30 errors ( most of them are due to unknown types)

Project build log:
```
In file included from ../MODBUS-LIB/Src/Modbus.c:14:
C:/Users/petrikas.lu/Desktop/WORK/UTB/MODBUS-LIB/Inc/Modbus.h:195:2: error: unknown type name 'osMessageQueueId_t'
  195 |  osMessageQueueId_t QueueTelegramHandle;
      |  ^~~~~~~~~~~~~~~~~~
C:/Users/petrikas.lu/Desktop/WORK/UTB/MODBUS-LIB/Inc/Modbus.h:198:2: error: unknown type name 'osThreadId_t'
  198 |  osThreadId_t myTaskModbusAHandle;
      |  ^~~~~~~~~~~~
C:/Users/petrikas.lu/Desktop/WORK/UTB/MODBUS-LIB/Inc/Modbus.h:204:2: error: unknown type name 'osSemaphoreId_t'
  204 |  osSemaphoreId_t ModBusSphrHandle;
      |  ^~~~~~~~~~~~~~~
../MODBUS-LIB/Src/Modbus.c:44:7: error: unknown type name 'osMessageQueueAttr_t'
   44 | const osMessageQueueAttr_t QueueTelegram_attributes = {
      |       ^~~~~~~~~~~~~~~~~~~~
../MODBUS-LIB/Src/Modbus.c:45:8: error: field name not in record or union initializer
   45 |        .name = "QueueModbusTelegram"
      |        ^
../MODBUS-LIB/Src/Modbus.c:45:8: note: (near initialization for 'QueueTelegram_attributes')
../MODBUS-LIB/Src/Modbus.c:45:16: warning: initialization of 'int' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
   45 |        .name = "QueueModbusTelegram"
      |                ^~~~~~~~~~~~~~~~~~~~~
../MODBUS-LIB/Src/Modbus.c:45:16: note: (near initialization for 'QueueTelegram_attributes')
../MODBUS-LIB/Src/Modbus.c:49:7: error: unknown type name 'osThreadAttr_t'
   49 | const osThreadAttr_t myTaskModbusA_attributes = {
      |       ^~~~~~~~~~~~~~
../MODBUS-LIB/Src/Modbus.c:50:5: error: field name not in record or union initializer
   50 |     .name = "TaskModbusSlave",
      |     ^
../MODBUS-LIB/Src/Modbus.c:50:5: note: (near initialization for 'myTaskModbusA_attributes')
../MODBUS-LIB/Src/Modbus.c:50:13: warning: initialization of 'int' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
   50 |     .name = "TaskModbusSlave",
      |             ^~~~~~~~~~~~~~~~~
../MODBUS-LIB/Src/Modbus.c:50:13: note: (near initialization for 'myTaskModbusA_attributes')
../MODBUS-LIB/Src/Modbus.c:51:5: error: field name not in record or union initializer
   51 |     .priority = (osPriority_t) osPriorityNormal,
      |     ^
../MODBUS-LIB/Src/Modbus.c:51:5: note: (near initialization for 'myTaskModbusA_attributes')
../MODBUS-LIB/Src/Modbus.c:51:18: error: 'osPriority_t' undeclared here (not in a function); did you mean 'osPriority'?
   51 |     .priority = (osPriority_t) osPriorityNormal,
      |                  ^~~~~~~~~~~~
      |                  osPriority
../MODBUS-LIB/Src/Modbus.c:51:17: warning: excess elements in scalar initializer
   51 |     .priority = (osPriority_t) osPriorityNormal,
      |                 ^
../MODBUS-LIB/Src/Modbus.c:51:17: note: (near initialization for 'myTaskModbusA_attributes')
../MODBUS-LIB/Src/Modbus.c:51:32: error: expected '}' before 'osPriorityNormal'
   51 |     .priority = (osPriority_t) osPriorityNormal,
      |                                ^~~~~~~~~~~~~~~~
../MODBUS-LIB/Src/Modbus.c:49:49: note: to match this '{'
   49 | const osThreadAttr_t myTaskModbusA_attributes = {
      |                                                 ^
../MODBUS-LIB/Src/Modbus.c:55:7: error: unknown type name 'osThreadAttr_t'
   55 | const osThreadAttr_t myTaskModbusA_attributesTCP = {
      |       ^~~~~~~~~~~~~~
../MODBUS-LIB/Src/Modbus.c:56:5: error: field name not in record or union initializer
   56 |     .name = "TaskModbusSlave",
      |     ^
../MODBUS-LIB/Src/Modbus.c:56:5: note: (near initialization for 'myTaskModbusA_attributesTCP')
[... 213 lines cut ...]
In file included from ../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.h:59,
                 from ../MODBUS-LIB/Src/Modbus.c:10:
../Middlewares/Third_Party/FreeRTOS/Source/include/queue.h:868:41: note: expected 'QueueHandle_t' {aka 'struct QueueDefinition *'} but argument is of type 'int'
  868 | BaseType_t xQueueReceive( QueueHandle_t xQueue, void * const pvBuffer, TickType_t xTicksToWait ) PRIVILEGED_FUNCTION;
      |                           ~~~~~~~~~~~~~~^~~~~~
In file included from ../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.h:60,
                 from ../MODBUS-LIB/Src/Modbus.c:10:
../MODBUS-LIB/Src/Modbus.c:1150:23: warning: passing argument 1 of 'xQueueSemaphoreTake' makes pointer from integer without a cast [-Wint-conversion]
 1150 |    xSemaphoreTake(modH->ModBusSphrHandle , portMAX_DELAY); //before processing the message get the semaphore
../Middlewares/Third_Party/FreeRTOS/Source/include/semphr.h:289:74: note: in definition of macro 'xSemaphoreTake'
  289 | #define xSemaphoreTake( xSemaphore, xBlockTime )  xQueueSemaphoreTake( ( xSemaphore ), ( xBlockTime ) )
      |                                                                          ^~~~~~~~~~
In file included from ../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.h:59,
                 from ../MODBUS-LIB/Src/Modbus.c:10:
../Middlewares/Third_Party/FreeRTOS/Source/include/queue.h:1418:47: note: expected 'QueueHandle_t' {aka 'struct QueueDefinition *'} but argument is of type 'int'
 1418 | BaseType_t xQueueSemaphoreTake( QueueHandle_t xQueue, TickType_t xTicksToWait ) PRIVILEGED_FUNCTION;
      |                                 ~~~~~~~~~~~~~~^~~~~~
make: *** [MODBUS-LIB/Src/subdir.mk:22: MODBUS-LIB/Src/UARTCallback.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [MODBUS-LIB/Src/subdir.mk:22: MODBUS-LIB/Src/Modbus.o] Error 1
"make -j16 all" terminated with exit code 2. Build might be incomplete.
```

For example, in Modbus.c, multiple types are not recognised:
[image]

krupis, 2022-08-10:
> Hi, You also need to enable FreeRTOS CMSIS_v2 under the middleware section in CubeMX. I updated the README to reflect this requirement.

Thanks for the fast response. I have enabled CMSIS_v2 but I have some other issues. I dont understand why the files are failing to link to each other. For example:
In Modbus.c it says that modbusHandler_t is unknown type:
[image]

Even though I can see that it is defined in Modbus.h:
[image]

And I include this header in Modbus.c

krupis, 2022-08-10:
Thats great. I have managed to build project with your library. I have a few questions additional questions:
1. If I want to use DMA for modbus RTU. Do I have to use circular or normal mode? At the moment I have both RX and TX DMA requests on circular modes.

[image]

2. I tried to look at examples that you provide for various microcontrollers but it is still not fully clear to me how can I write some data to the registers as modbus master.
For example, I want to write to slave 1, holding register offset 5 value 0x1234. Could you show me an example what function I need to call to do that. Thanks in advance!

alejoseb, 2022-08-11:
The DMA configuration does not require the circular mode. Check the corresponding examples.

Also check the following file, it contains an example for the modbus master 

Examples/ModbusF429/Core/Src/freertos.c

krupis, 2022-08-11:
Thats great.  I used the example that you have given to read slave register:
```
void StartDefaultTask(void *argument)
{
  /* USER CODE BEGIN 5 */
  /* Infinite loop */
	uint32_t u32NotificationValue;

	telegram.u8id = 1; // slave address
	telegram.u8fct = 3; // function code (this one is registers read)
	//telegram[0].u16RegAdd = 0x160; // start address in slave
	telegram.u16RegAdd = 0x0; // start address in slave
	telegram.u16CoilsNo = 1; // number of elements (coils or registers) to read
	telegram.u16reg = ModbusDATARX; // pointer to a memory array
  /* Infinite loop */
  for(;;)
  {
	  ModbusQuery(&ModbusH, telegram); // make a query
	  u32NotificationValue = ulTaskNotifyTake(pdTRUE, 500); // block until query finishes or timeout
	  if(u32NotificationValue)
	  {
		  DEBUG_PRINT("modbus responded \n");
		//handle error
		//  while(1);
	  }
	  osDelay(500);

  }
  /* USER CODE END 5 */
}
```

I can see on the logic analyzer that the data has been requested and sent.

I tried to debug what is happening in the code line by line but it seems very complicated and there are multiple layers behind this (Multiple freertos tasks, multiple queues and etc...)

I have been able to confirm that whenever I send data, SendQuery is getting executed. What I am confused about is the following:
[image]

When I step the code with breakpoints, I can see that the case case MB_FC_READ_INPUT_REGISTER: is activated. I dont understand why that happens since I send modbus request with funcion code 3 so case MB_FC_READ_REGISTERS: should be activated (Not sure why its empty)

Another question that I have:
Do you have a modbus parser implemented to print messages console? I mean when I send command to read certain register. For example I want to read multiple slave 1 coil status with address offset 5. Is there any way to printf result to the console for example:
[SLAVE1] COIL5 = 0
[SLAVE1] COIL6 = 1
[SLAVE1[ COIL7 = 0

At the moment, I can see that my commands are working and I can send requests to read various registers but there is no point reading anything if I cant see the output.

Im sorry for troubling you so much

At the m

alejoseb, 2022-08-11:
Hi,
MB_FC_READ_REGISTERS is not empty, it is handled by the code shown in line #870 of the code.

There is no functions to print to the console. In the example, you can check the data coming from the slave in the ModbusDATARX
buffer, you can use breakpoints to verify that or print to the console by yourself.

closed 2022-08-19 by alejoseb