alejoseb/Modbus-STM32-HAL-FreeRTOS #59: Any getting started documentation?

IanAber, 2022-09-16:
Is there any documentation on how to get started? I need to configure a modbus device that will be a slave and return things like GPIO states and Analogue read values in coils and registers. It would be great if there were a simple guide to providing coils, holding registers and input registers for example. Also a simple guide on how to configure the UART port that is used including the pin for driving TXEN on the RS485 driver.
Did I miss something obvious. I am not a trained programmer and I have no idea where to look to get this info if is not in a readme or similar.

alejoseb, 2022-09-17:
Besides the Readme file in the repository there is no more documentation available.

The Blue Pill  example already does something similar to what you need using the USART1. 

You can see how data coming from Modbus is used to turn ON or OFF a GPIO in the following line:

https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/742f6c8d3cc56bcbcf8245c626e2f4517ebf102b/Examples/ModbusBluepill/Core/Src/freertos.c#L152

This library uses a single memory space for coils, holding registers and so on. So, its up to you which addresses are used for each one. The address space starts at 0 and the upper limit is the size of the array (ModbusDATA) that you need to configure when the library is initialized.

You can see the initialization code here:

https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/742f6c8d3cc56bcbcf8245c626e2f4517ebf102b/Examples/ModbusBluepill/Core/Src/main.c#L94

The readme file provides the instructions to configure the serial port. I recommend using the CubeMX tool included in the STM32CubeIDE. All the examples are pre-configured projects that works directly with the STM32CubeIDE and CubeMX.
Also, to control the RAS485 TXEN you need to provide a pre-configured GPIO as part of the initialization. So first configure 
the GPIO as an output, and then use it in the initialization of the Modbus library. 

Check this example for the initialization:

https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/742f6c8d3cc56bcbcf8245c626e2f4517ebf102b/Examples/ModbusF103/Core/Src/main.c#L101
  
Hope this helps.

IanAber, 2022-09-17:
Many, many thanks. It looks like this is what I need to get my head round it. I really appreciate the effort.
Ian Abercrombie

***@***.*** ***@***.***>
Phone: +1 (352) 281-6286
www.CedarTechnology.com <http://www.cedartechnology.com/>

closed 2022-09-17 by alejoseb