alejoseb/Modbus-STM32-HAL-FreeRTOS #60: Numbering of registers

IanAber, 2022-09-21:
Am I correct in thinking that, assuming the STM32 is a slave and implementing 5 coils, 5 holding registers, 5 input registers and 5 discretes, the numbering for each has to be unique? i.e. Coils might be 0 to 4, holding registers 5 to 9 etc. You cannot have coil 0 AND holding register 0 at they would use the same value.
Also, if you decide the holding registers are 5 to 9 there is nothing to stop the master from requesting, or writing to holding register 0 thereby actually reading or writing a coil even though a holding register action was specified.

alejoseb, 2022-09-21:
The addressing scheme is a bit different to what you described. Coils are stored as bits of a 16bit register. For example, coils with addresses between 0 and 15 will be sharing the same memory space of the holding register (i.e., a 16 bit variable) at address 0. So on and so forth. 

There is nothing preventing the master to access a particular address as a coil or as a holding register.

IanAber, 2022-09-29:
Is there a standard way to return an error if the requested coil or register does not exist? I see the return from the functions provides the buffer size but there does not seem to be an obvious way to show that an error occurred.

alejoseb, 2022-09-30:
The slave handles the errors according to Modbus specification. What you need is not part of the Modbus protocol.

closed 2022-10-11 by alejoseb