alejoseb/Modbus-STM32-HAL-FreeRTOS #62: Fabulous Modbus Library

ozdemirmcihad, 2022-11-15:
First of all, thank you for providing this service to us. I'm going to ask you a few questions about your library.

1-)t35 is set to =5, this is why 5 . Do we need to change at different baud rates?
2-) I have never seen other errors occur in the system except for the CRC error. For example, which error flag should be activated when the communication is interrupted, how can we try this?
3-) Does the system support 2 usart lines to work as slaves at the same time as a multiusart or will there be any confusion?
4-) Finally, how can we use the system outside of FreRtos.

Thank you in advance for your answers. If you do not want to answer here, we can discuss these issues via whatsapp or mail.
Thank you again.

alejoseb, 2022-11-15:
1-)t35 is set to =5, this is why 5 . Do we need to change at different baud rates?
That is part of the Modbus standard. You don't need to change the baudrate, instead for some baudrates or when connecting to devices that are not Modbus compliant it might be necessary to adjust the t35 parameter, you can let it as is.

2-) I have never seen other errors occur in the system except for the CRC error. For example, which error flag should be activated when the communication is interrupted, how can we try this?
Modbus is a pretty simple protocol. Besides CRC, overflows and function codes not supported, there is no other type of error handled or implemented in this library. 

3-) Does the system support 2 usart lines to work as slaves at the same time as a multiusart or will there be any confusion?
yes it supports multiple slaves or masters and combinations 

 4-) Finally, how can we use the system outside of FreRtos.
This library is specifically designed to work with FreeRTOS you can look around there are plenty of simpler implementations that can be run without FreeRTOS

I do not provide personal support by any means. The library is provided as is, you can check the license as well for more details. I will only answer general questions here (i.e., github issues).

ozdemirmcihad, 2022-11-16:
First of all, thank you very much for your answers.
With your permission, I have 2 more questions.
1-) Modbus is a very simple protocol. CRC, overflows, and function codes are not supported in this library, and there are no other types of errors handled or implemented.

What do you mean here, I'm a little confused. Did you mean that this library does not catch these errors or did you mean that this library does not fall into such errors?

2-) I was wondering if the codes in this library were prepared specifically for industrial applications, that is, if they were sufficiently tested. Do you have a long-term test history?

Thank you .

alejoseb, 2022-11-16:
1) These are the three types of controls that are implemented:
CRC, overflows or data length checks, and validations of function codes. Anything else is either not part of the Modbus protocol or not implemented.

2) The code is delivered "as is", it has been used extensible in various products but without any implicit warranty,
please check the license file https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/master/LICENSE

ozdemirmcihad, 2022-11-16:
By the way, I forgot to ask the thing. T35=5 is it 5 ms here or us or seconds?

alejoseb, 2022-11-17:
It is "ticks" of the FreeRTOS which for the standard configuration is 5 ms

ozdemirmcihad, 2022-11-29:
Hi bro sorry for bothering you.

I want to use multiple slaves in Modbus library.

ModbusH.uModbusType = MB_SLAVE;
ModbusH.port = &huart3;
ModbusH.u8id = 1;
ModbusH.u16timeOut = 50;
ModbusH.EN_Port = NULL;
ModbusH.u16regs = ServerBuf3;
ModbusH.u16regsize= sizeof(ServerBuf3)/sizeof(ServerBuf3[0]);
ModbusH.xTypeHW = USART_HW;

//Initialize Modbus library
ModbusInit(&ModbusH);
//Start capturing traffic on serial Port
ModbusStart(&ModbusH);

ModbusH.uModbusType = MB_SLAVE;
ModbusH.port = &huart3;
ModbusH.u8id = 2;
ModbusH.u16timeOut = 50; // this time is important to avoid crc error
ModbusH.EN_Port = NULL;
ModbusH.u16regs = ServerBuf3;
ModbusH.u16regsize= sizeof(ServerBuf3)/sizeof(ServerBuf3[0]);
ModbusH.xTypeHW = USART_HW;

//Initialize Modbus library
ModbusInit(&ModbusH);
//Start capturing traffic on serial Port
ModbusStart(&ModbusH);

When I make two configurations in this way, the system hangs. Is my method wrong for multiple slaves?

alejoseb, 2022-11-29:
yes it is wrong you are using the same huart3 and buffers for the 2 slaves. Each slave or master needs its own serial port and buffers. Check the examples in the repository.

ozdemirmcihad, 2022-11-29:
Thank you for your answer . Actually, I have 2 f446 cards in my hand. I'm trying to communicate with each other using your library.

Two cards communicate one by one with modbus poll. However, I was confused in their communication with each other.

1-) Should I choose a card as a master?
2-)
ModbusH.uModbusType = MB_SLAVE;
ModbusH.port = &huart3;
ModbusH.u8id = 1;
ModbusH.u16timeOut = 50; // this time is important to avoid crc error
ModbusH.EN_Port = NULL;
ModbusH.u16regs = ServerBuf3;
ModbusH.u16regsize= sizeof(ServerBuf3)/sizeof(ServerBuf3[0]);
ModbusH.xTypeHW = USART_HW;

//Initialize Modbus library
ModbusInit(&ModbusH);
//Start capturing traffic on serial Port
ModbusStart(&ModbusH);

ModbusH1.uModbusType = MB_MASTER;
ModbusH1.port = &huart1;
ModbusH1.u8id = 0;
ModbusH1.u16timeOut = 50;
ModbusH1.EN_Port = GPIOB;
ModbusH1.EN_Pin = GPIO_PIN_0;

ModbusH1.u16regs = ServerBuf;
ModbusH1.u16regsize= sizeof(ServerBuf)/sizeof(ServerBuf[0]);
ModbusH1.xTypeHW = USART_HW;

//Initialize Modbus library
ModbusInit(&ModbusH1);
//Start capturing traffic on serial Port
ModbusStart(&ModbusH1);

Is it enough to just do this part of the code or do I need to call another function for master?

Thank you.

alejoseb, 2022-11-29:
Modbus uses the master-slave paradigm, so the communication is always started by the master who sends a request, which is later answered by the slave. Notice that a serial port can only be used for master or slave, they are mutually exclusive. 
Check the examples for using the master profile.

Aimhighn, 2022-12-01:
Hi alejoseb,
Thank you for the codes you posted.
I used example ModbusF429TCP for my board. My board is STM32F429IGT6. My board with IP 192.168.1.190 is the client or master. I have used Modbus Slave software as a server or slave in the system with IP 192.168.1.103. Currently, the problem I have is that the connection is disconnected after a short period of time. I am attaching the photos of Wireshark.
[image]
[image]
[image]
[image]