alejoseb/Modbus-STM32-HAL-FreeRTOS #64: TX GERİ ARAMA ÇALIŞMAMA SORUNU

ozdemirmcihad, 2022-12-21:
Hello;
I ran your code and there doesn't appear to be a problem. But I have a code that returns another task in FreeRTOS, and this code checks a state with a while loop. TxCalback does not work when the state controlled by this while loop is stuck in while. RXCALBACK works but. Can you help me with this? Shouldn't txcallback be running all the time?
It does not go into the switch structure with the function codes, I am waiting for your help when the code is inserted.

ozdemirmcihad, 2022-12-21:
void alt_klepe_open(int klepe_wait)
{
	if(	controlklepe.ust_klepe_status==0 ||controlflowparameters.baypass==1 || ComunucationBuf4==1 ||controlflag.baypass_manuel1==1|| controlcumuparameters.baypass_oled==1)//üst kelepe kapalıysa ve bayas yoksa
	{
		HAL_GPIO_WritePin(GPIOC,ALTKLEPEPININ,GPIO_PIN_SET ); //alt kelepyi aç
		while(controlklepe.alt_klepe_status!=0) //alt kelepe açık değilse açılmayı beklemek için whileda bekle
		{
			controlklepe.alt_klp_open_timer_start=1; //alt klepe hatasını saydırmak için bayragı kaldır.
			controlcount.alt_klp_sayac_start_open=1;
			if(ComunucationBuf1==0 )//|| controlflowparameters.start_status==0) //alt klepe acilana kadar bekle acilinca burada bak kapandıysa diğer işleme geç
			{
				if(controlcount.alt_klp_sayac_open>=klepe_wait)
				{
					controlklepe.alt_klepe_status=0;
					controlklepe.alt_klp_open_timer_start=0;
					controlcount.alt_klp_sayac_start_open=0;
				}
			}
			if( controlflowparameters.baypass==1 || ComunucationBuf4==1|| controlflag.baypass_manuel1==1 || controlcumuparameters.baypass_oled==1 || controlflowparameters.start==0)//baypass ve stop durumunda whileden çıkması için
			{
				controlklepe.alt_klepe_status=0;
				controlklepe.alt_klp_open_timer_start=0; //timerleri sıfırla
				controlcount.alt_klp_sayac_start_open=0;
			}
		}
	}
}

When I enter while inside this structure in another task, txcallbacte does not enter. How can I solve this?

alejoseb, 2022-12-21:
Hi, 
Sorry, this is something beyond of what I can help with, and not related to the Modbus library at all.
Try to debug your code before integrating with Modbus so you can isolate your issue.
Good luck!

closed 2022-12-21 by alejoseb

ozdemirmcihad, 2022-12-21:
my question is actually directly related to modbus. What I mean is this. Normally TX or rx CALLBACK FUNCTIONS are not affected by loops like while. But here tx does not enter the callback function.
My question to you is how do you trigger the tx callback function

alejoseb, 2022-12-22:
Your question is not related to the modbus protocol, since the callbacks are particular to each implementation and not to the protocol. However, this library uses the callbacks for the TX and RX events of USART as defined defined in the file: 

https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/master/MODBUS-LIB/Src/UARTCallback.c 

You can add your own code there in the corresponding placeholders. Check the comments.

Besides, using a while loop may consume the whole processing time of your microcontroller.  Since you are using FreeRTOS you must implement a method based on semaphores or signals that yields execution of the loop and executes other tasks .  These concepts of multitasking are beyond the context of this library and you must check FreeRTOS documentation.