alejoseb/Modbus-STM32-HAL-FreeRTOS #65: Slave DMA connection

DmitryBruckhanov, 2023-02-13:
I am at the very beginning of the STM journey and cannot figure out how to connect your library in DMA slave mode, indicating my addresses and data in them. help me please

alejoseb, 2023-02-15:
I am not sure what are you asking for. The DMA mode is just a specific operation mode of the USART. Besides, the usage of the library is the same. So I recommend you check any of the examples without DMA first.

DmitryBruckhanov, 2023-02-15:
I have no problems connecting your library. I'm having trouble filling in data by address. I have not worked with a port other than 232 or 485 (modbus) before. If possible, then some example of how to fill in the data for the Master. Sorry, I'm using google translate

alejoseb, 2023-02-19:
Sorry I cannot help here since your question is not clear. Please provide more details of what are you trying to do and what is your configuration.

DmitryBruckhanov, 2023-02-21:
Good afternoon.  Is it possible to pass strings in your library?  I am asking for an example of how to pass data.  Your examples are too light for a real project.

alejoseb, 2023-02-21:
If you need a protocol to send arbitrary data (such as strings) by design you are in the wrong place. This is an implementation of Modbus, i.e., an industrial protocol meant to work with PLCs data types e.g, coils and registers.  

Nevertheless, you can overload the protocol to send strings combining several contiguous registers and manipulating pointers. That is not part of the Modbus specification, thus not implemented and not showcased by any example.

closed 2023-02-21 by DmitryBruckhanov