alejoseb/Modbus-STM32-HAL-FreeRTOS #66: Use TCP slave without UART

alejandroalis, 2023-02-25:
Sorry this is a question, not an issue, but don't know where to ask...

One question, I'm trying to use the H743TCP example on a H735 device but only want to use the TCP protocol and It seems that the USART is always enabled, it's possible to use the library without any uart?

Thanks!

alejoseb, 2023-02-25:
I think the library requires some headers and references to USART even when not used. Solving that might require some cleaning of the library. Meanwhile you can add a serial port to your project and simply don't use it.

alejandroalis, 2023-02-25:
yes, working without uart because I don't setup it and TCP is working.. I'm trying to understand how the library works because I never used modbus and very confused about the registers (coils, inputs..)

closed 2023-03-06 by alejoseb