alejoseb/Modbus-STM32-HAL-FreeRTOS #70: How to write code multiple holding register

KomKGT, 2023-05-22:
Hello @alejoseb your library is well but I stuck about write code multiple holding register ,Can you give me an example Thank you.I add my code in picture 
[image]
it not work. What's wrong in my Code?

alejoseb, 2023-05-24:
Hi, Check this example https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/e9c16cf5a3e8fceb9ff06124b63ecf8fef9dc66c/Examples/ModbusF429/Core/Src/freertos.c#L120

It is for reading multiple registers, but the idea for writing is exactly the same

KomKGT, 2023-05-27:
Thank you sir. I got it 
[image]
This is my code it's work.

closed 2023-05-27 by KomKGT

alejoseb, 2023-05-27:
I don't quite understand what you did, but I am glad it works for you.