alejoseb/Modbus-STM32-HAL-FreeRTOS #71: Not really an issue...

dolence, 2023-05-23:
Hi @alejoseb

I've been testing your library for a while and it is really nice what you did. Thank you!
I have a question and I think you would be the person to point me into the right direction. I have a situation where I need to read many sensors using RS-485. I will have a maximum of 8 cables with up to 32 sensors on each cable. Master send a request to convert temperature of one cable and slave should answer  with that cable array of sensors data. My understanding is that Modbus holds data in registers and master read each of these registers one by one or a sequence of registers. Reserving one registers for each sensor would consume much resources. Is there any smarter way to do this than using a simpler and custom RS-485 protocol where master could send commands like "convert sensor from cable 1" or "convert sensor 10 from cable 2", parse this command on slave and send the data back? At first modbus-rtu seemed to be a nice fit and your library is freertos aware and support DMA, but then things become tricky. What you think?

Thank you in advance.

alejoseb, 2023-05-25:
Your idea should work without any issue.
Another option could be to extend the library with a callback right after the slave parsed the command https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/e9c16cf5a3e8fceb9ff06124b63ecf8fef9dc66c/MODBUS-LIB/Src/Modbus.c#L732 You send a first request or command to read the sensors and only respond after having all the data ready in the holding registers, then you issue a new request to read the data from them. It is a more complex solution and only works if reading from sensors takes just a few milliseconds.

dolence, 2023-05-26:
Hi, thank you for taking your time to help. I think your advice really worth a try. Sensors conversion time at the maximum resolution is something around 750 ms, but I think I can send a command to covert all the connected sensors at same time and just read the sensor register containing the temperature after that time has passed. No need to convert and wait one by one.

closed 2023-06-15 by alejoseb