alejoseb/Modbus-STM32-HAL-FreeRTOS #72: Connect to multiple servers....

Aimhighn, 2023-05-27:
Hello. Hope you're well.
I solved the problem by changing your library code from CMSIS V2 to CMSIS V1.
Now I want to simultaneously write and read from one client board to several servers.
For example, here are two servers with
Server1: IP:192.168.1.101, Port:2000
Server2: IP:192.168.1.102, Port:2001
According to the picture, I considered a telegram for each of them.
In the case that both servers are ready to accept and communicate, there is no problem and data exchange is done, but when one server fails, the connection with the other server also fails.
[image]
[image]
How can I solve this problem?

alejoseb, 2023-07-11:
I am not sure what could be the reason, I need to replicate this configuration to verify the issue.

GROMKABAN, 2023-09-22:
Same problem. Found a solution?

TheOrignalCDV, 2023-10-26:
Hi Aimhighn,
A little off-topic but I note that in issue #63 that you successfully converted the library to CMSIS V1.  

I have been struggling to get this going for a few weeks now. Are you able to assist with a branch to your V1 library?

Thanks and best regards,
The OriginalCDV

Aimhighn, 2023-10-29:
> Same problem. Found a solution?
Hi. I almost know where the problem is. But I don't have time to modify the library code. If I can, within the next 10 days, I will send the correction code for this work.

Aimhighn, 2023-10-29:
> Hi Aimhighn, A little off-topic but I note that in issue #63 that you successfully converted the library to CMSIS V1.
> 
> I have been struggling to get this going for a few weeks now. Are you able to assist with a branch to your V1 library?
> 
> Thanks and best regards, The OriginalCDV

Hi. Yes. I have already succeeded in porting the library from CMSIS-V2.0 to CMSIS-V1.0. Please wait while we create a fork from the modified code.

Benedito821, 2025-07-04:
A bit out off topic: does this library use sockets?