alejoseb/Modbus-STM32-HAL-FreeRTOS #77: Task is suspended forever

marnidRed, 2023-07-11:
Hello,

I'm using your library with FreeRTOS. I configured several tasks - two of them are responsible for MODBUS-RTU communication: just two independent masters on two USARTs. Unfortunetly always after some hours of working, one of these two tasks get state "suspended" forever. "Suspended" is also task "TaskModbusMaster". Have you any idea, what's wrong? Of course I'll give more informations about system, but please tell me what is necesesery.

marnidRed, 2023-07-12:
Thank you for the answer. I see that that my problem in modbus.c is here - I never get notification, but I have to debug why.
[image]

alejoseb, 2023-07-13:
You have to verify that the timeout timer is actually running. If you run-out of memory the timeout will be also suspended and, if the slave never answer the timeout will not be triggered and you will end up with the task suspended indefinitely.

closed 2023-08-03 by alejoseb