alejoseb/Modbus-STM32-HAL-FreeRTOS #78: modbus

ozdemirmcihad, 2023-07-13:
Hello, with your permission, I have two questions for you.
1-) I tried your library stm32g473 cmsis 2 but it didn't work. Does the code not support this processor?

2-) In another question, uart interrupts in freertosta affect my main task and the displays are flickering, what can I do about the solution?

Thank you very much in advance.

alejoseb, 2023-07-13:
1) The library should work with any stm32 microcontroller, eventually it needs small adjustments here: https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/d85e0bc5858f71534f690fae837c14378817ad0f/MODBUS-LIB/Src/Modbus.c#L1572 
to support a specific MCU.

2) That question is beyond the scope of this library. However, I can recommend using the DMA mode, which uses less MCU cycles.

ozdemirmcihad, 2023-07-14:
Thank you for your answer
-> #if defined(STM32H7) || defined(STM32F3) || defined(STM32L4) || defined(STM32L082xx)
It is enough to add stm32g473 statement to this line, is it correct?

ozdemirmcihad, 2023-07-14:
Although I added g4 here, the system did not work. RXCALLBACK works, but the code does not enter TxCallback at all. What could be the reason?

ozdemirmcihad, 2023-07-14:
I am running two tasks. I have display function in other task. When the display function works, the communication hangs, what do you think might be the reason?

alejoseb, 2023-07-14:
It is difficult to know the reason, but you have to architect your tasks for being compatible with a multi-thread environment. This means that you need to avoid  spins (while loops) waiting for specific events for the LCD. Instead you need to use primitives to yield the execution of the LCD task to let the Modbus Task to run while the LCD is not doing anything. You also need to check the priority of each task. A high priority task will block a low priority one. All of these details are beyond the scope of this library and I cannot provide support.

ozdemirmcihad, 2023-07-14:
thank  you

closed 2023-07-17 by alejoseb