alejoseb/Modbus-STM32-HAL-FreeRTOS #81: freertos crashes after changing baudrate

ozdemirmcihad, 2023-08-03:
Hello;
Communication works fine with baud value of 115200 at first boot. Later I want to change the baudrate to 9600 while working. When I do this, that task of the freertos freezes or the timer freezes.

This is the code I changed baud.
if(flaggg==1)
		{
			MX_USART1_UART_Init(9600);
			//Initialize Modbus library
			ModbusH.uModbusType = MB_SLAVE;
			ModbusH.port =  &huart1;
			ModbusH.u8id = 1;
			ModbusH.u16timeOut = 100; // bu zaman crc hatasına düşmemesi için önemli
			ModbusH.EN_Port = NULL;
			ModbusH.u16regs = ServerBuf1;
			ModbusH.u16regsize= sizeof(ServerBuf1)/sizeof(ServerBuf1[0]);
			ModbusH.xTypeHW = USART_HW;
				ModbusInit(&ModbusH);
				//Start capturing traffic on serial Port
				ModbusStart(&ModbusH);
		}
		if(flaggg==2)
			{
			MX_USART1_UART_Init(115200);
			//Initialize Modbus library

			ModbusH.uModbusType = MB_SLAVE;
			ModbusH.port =  &huart1;
			ModbusH.u8id = 1;
			ModbusH.u16timeOut = 100; // bu zaman crc hatasına düşmemesi için önemli
			ModbusH.EN_Port = NULL;
			ModbusH.u16regs = ServerBuf1;
			ModbusH.u16regsize= sizeof(ServerBuf1)/sizeof(ServerBuf1[0]);
			ModbusH.xTypeHW = USART_HW;
				ModbusInit(&ModbusH);
				//Start capturing traffic on serial Port
				ModbusStart(&ModbusH);
			}

The part where the code is stuck is in modbus.c.

		if  (modH->myTaskModbusAHandle == NULL)
		{
			while(1); //Error creating Modbus task, check heap and stack size
		}

it gets stuck in the while loop here. What should I do to keep it from getting stuck?

alejoseb, 2023-08-03:
The method that you are using looks wrong.

Calling ModbusInit(&ModbusH) creates a new FreeRTOS task to handle the modbus protocol. Therefore, that will break completely what you are doing, since 2 task will be associated with the same serial port.

Besides, you should stop and de-initialize the USART before changing parameters. After changing parameters, you should only call ModbusStart(&ModbusH);. 

As I said, this is not supported by default, so I cannot assure if my recommendation works, you might need more changes.

ozdemirmcihad, 2023-08-03:
if(flaggg==1)
			{
				MX_USART1_UART_Init(9600);

					ModbusStart(&ModbusH);
			}
			if(flaggg==2)
				{
				HAL_UART_MspDeInit(&huart1);

				MX_USART1_UART_Init(115200);
				//Initialize Modbus library
				ModbusStart(&ModbusH);
				}

yes i did it that way but it still didn't work again.

ozdemirmcihad, 2023-08-03:
//check that port is initialized
while (HAL_UART_GetState(modH->port) != HAL_UART_STATE_READY)
{

}

stuck on this part

closed 2023-08-03 by ozdemirmcihad