alejoseb/Modbus-STM32-HAL-FreeRTOS #83: error: #20: identifier "UART_HandleTypeDef" is undefined， error: #20: identifier "GPIO_TypeDef" is undefined

Eric-Hsia, 2023-09-06:
hi, eziya, when I compile ModbusF429 example by keil mdk, there has undefined identifier errors, from the modbus.h headfile, it seems didn't include these two indentifier, thanks.
[image]

Eric-Hsia, 2023-09-06:
hi,  when I compile ModbusF429 example by keil mdk, there has undefined identifier errors, from the modbus.h headfile, it seems didn't include these two indentifier, thanks.
[image]

alejoseb, 2023-09-08:
Hi, 
You need to update the include links accordingly. This is not a library issue, instead a configuration issue on your IDE.

closed 2023-09-08 by alejoseb