alejoseb/Modbus-STM32-HAL-FreeRTOS #86: Documentation

vishnu-illikkal, 2023-09-19:
I need more documentation on how to set up and run this library in STM32. 
I am trying to work with a modbus peacefair pzem-016 energy meter.
Anyhow I need to create a documentation for how to do this. 
But I can't find any on the internet.
So if you can explain how the library works I can create a HTML page and share in this repo.

1 : Give me details on how this library works 
2 : I will write a documentation on HOW-TO in HTML
  
Thank you for your time.

alejoseb, 2023-09-19:
Hi all the documentation is in the readme file of this repository. 
If you are willing to extend the documentation it has to be written in markdown, and posted in this repository.
Let me know if you have any specific questions regarding the content of the readme file

alejoseb, 2023-10-10:
hi @j-broome , that is not an easy question since the TCP stack can be implemented in different. In this library ,in particular, I took advantage of the STM32 HAL library, which uses the LWIP stack. In your case, the TCP stack is implemented by hardware, which is quite different. In summary, you should take a look to the function implemented here: https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/861d511588915796447184e1e384c11becfb1466/MODBUS-LIB/Src/Modbus.c#L486 . That function implements the main logic to handle TCP connections and an aging algorithm that I designed to close connections. Besides that I cannot help much here, since that is outside the scope of this library.

alejoseb, 2023-10-10:
Modbus over TCP is not exactly the same, frames have small differences. This version of the library cope with those differences, when TCP is activated. So better you check Modbus documentation and implement accordingly. Besides that, the library in this repo is equivalent to the Pico port. And yes TCP was left out on purpose since the original Pico does not support TCP natively.

closed 2023-10-11 by alejoseb