alejoseb/Modbus-STM32-HAL-FreeRTOS #87: FreeRTOS CMIS-V1

TheOrignalCDV, 2023-09-27:
Firstly, this is a terrific library and the effort on this library and thank you for your effort.

The issue I would like to bring is NOT related to your code specifically. However there is a known bug with the STM32 CubeMX LWIP library when used with FreeRTOS. Specifically when used with CMIS-V2.

This has been documented on a few boards, however the ST site probably has the best information on it:
https://community.st.com/t5/stm32cubemx-mcus/bug-in-cubemx-ide-lwip-freertos-on-nucleo-f429zi/td-p/295367

Basically the EthernetIF semaphore calls never return, effectively making the STM32 'hang' after a few minutes (or a few seconds in my experience).

Apparently the FreeRTOS CMIS-V1 works flawlessly.

So the issue/request: is there a version of Modbus-STM32-HAL-FreeRTOS that works with CMIS-V1.

Thank you again.

alejoseb, 2023-09-27:
Hi,
I am aware of this issue. Please take a look to this one https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/issues/63 you can contact the person that claimed ported the library to CMSIS-V1, maybe he can help you.
I can consider to have another branch for that, if someone is willing to port the library, but I will not downgrade this library to CMSIS-V1 since it is not developed anymore, and does not support the newer Cortex-M architectures.

TheOrignalCDV, 2023-09-27:
Thanks for the super-fast feedback!! I understand the reluctance with taking a step backwards to CMIS-V1.

If I have any success, I will let you know.

Again, thanks!

closed 2023-09-27 by TheOrignalCDV