alejoseb/Modbus-STM32-HAL-FreeRTOS #92: help

HasanMahdavi777, 2023-11-28:
Hello
I want to do a modbus master with freertos on the stm32f107vc board, but I don't understand your code, how to tell the program to read a register from another slave, with which function model
Thank you for helping me further

alejoseb, 2023-11-28:
Hi,
Please check the examples. For instance check [here]( https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/d4895ddf6ea74dbee5ec580bd580844509572ee1/Examples/ModbusF103/Core/Src/freertos.c#L156) for an example of something quite similar to what you need.
Notice that you first need to integrate the library to your code and initialize the library, check the main.c file for an example of the required configuration.

HasanMahdavi777, 2023-11-29:
I looked at the examples
I want to read from a slave
How should I set the slave id and register function and address in the program?
I did not see it in the examples

alejoseb, 2023-11-29:
Did you really read my previous message? 
it is pointing to what you need, you need to prepare telegrams and push them to the Modbus queue.

HasanMahdavi777, 2023-11-29:
yes i read
I am a newbie
Thank you, can you write the part of the code that you say I should adjust and cover the telegrams?

alejoseb, 2023-11-29:
[read line](https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/d4895ddf6ea74dbee5ec580bd580844509572ee1/Examples/ModbusF103/Core/Src/freertos.c#L164)

closed 2023-11-29 by alejoseb