alejoseb/Modbus-STM32-HAL-FreeRTOS #108: 启用RS485模式后需要在收发控制管脚置位操作前后加入一些延时，避免在总线上出现乱字符

piggyy, 2024-07-17:
改成如下代码即可      
  if (modH->EN_Port != NULL)
        {
            // enable transmitter, disable receiver to avoid echo on RS485 transceivers
            HAL_HalfDuplex_EnableTransmitter(modH->port);
            osDelay(1);
            HAL_GPIO_WritePin(modH->EN_Port, modH->EN_Pin, GPIO_PIN_SET);
            osDelay(1);
        }

alejoseb, 2024-07-17:
Sorry, I only speak English and Spanish. Please use a translator so I can help you.

piggyy, 2024-07-18:
In the sendTxBuffer function in the modbus.c file,       
 if (modH->EN_Port != NULL)
        {
            // enable transmitter, disable receiver to avoid echo on RS485 transceivers
            HAL_HalfDuplex_EnableTransmitter(modH->port);
            HAL_GPIO_WritePin(modH->EN_Port, modH->EN_Pin, GPIO_PIN_SET);
        }
It should be modified to:
        if (modH->EN_Port != NULL)
        {
            // enable transmitter, disable receiver to avoid echo on RS485 transceivers
            HAL_HalfDuplex_EnableTransmitter(modH->port);
            osDelay(1);
            HAL_GPIO_WritePin(modH->EN_Port, modH->EN_Pin, GPIO_PIN_SET);
            osDelay(1);
        }
Because between switching the half-duplex transmission mode and the operation of setting the transmission and reception pins, it is necessary to add some delays before and after the operation of setting the transmission and reception control pins, to avoid instability in the bus level caused by the switch of RS485 transmission mode, resulting in garbled characters on the bus.

alejoseb, 2024-07-19:
This is not necessary unless your transceivers and electrical environment are not stable, you need first check that.
If this change help you improving stability, then it is up to you to modify your local copy of the library, but I will not include that in the mainstream codebase.

closed 2024-07-19 by alejoseb