alejoseb/Modbus-STM32-HAL-FreeRTOS #110: problem in running modus master code for STM32F730r8t6

nirbhow, 2024-08-14:
Dear sir,
i'm trying to use your code for STM32F730R8T6. as modbus master initialized as follows:
/* Modbus Master initialization */
  ModbusH.uModbusType = MB_MASTER;
  ModbusH.port =  &huart2; // This is the UART port connected to STLINK in the NUCLEO F429
  ModbusH.u8id = 0; //slave ID
  ModbusH.u16timeOut = 1000;
  ModbusH.EN_Port = NULL; // No RS485
  //ModbusH2.EN_Port = LD2_GPIO_Port; // RS485 Enable
  //ModbusH2.EN_Pin = LD2_Pin; // RS485 Enable
  ModbusH.u16regs = ModbusDATA;
  ModbusH.u16regsize= sizeof(ModbusDATA)/sizeof(ModbusDATA[0]);
  ModbusH.xTypeHW = USART_HW;
  //Initialize Modbus library
  ModbusInit(&ModbusH);
  //Start capturing traffic on serial Port
  ModbusStart(&ModbusH);
  //query initilization
  telegram.u8id =1;

But i'm facing problem that the code always stuck at  function
xQueueReceive(modH->QueueTelegramHandle, &telegram, portMAX_DELAY);
in the modbus.c library.

when i check the status of USART2.TDR register it is empty.

My second question is 
how i have to generate a modbus query for slave devices., which function i have to call for modbus query.

alejoseb, 2024-08-14:
The behavior that you are observing is expected. The library will wait there for a telegram to be sent through Modbus as a master. 
To send a request as a master check the master example for the F103 MCU the example in the freertos.c within the core folder

nirbhow, 2024-08-15:
Thanks a lot for your advice,
I've checked the example in the freertos.c of F103MCU.
As per my understanding
1.  i've to use following section in function *"void
StartTaskModbusMaster(void *argument)" .*

 // telegram 0: read registers
  uint32_t u32NotificationValue;
  telegram[0].u8id = 1; // slave address
  telegram[0].u8fct = 3; // function code (this one is registers read)
  //telegram[0].u16RegAdd = 0x160; // start address in slave
  telegram[0].u16RegAdd = 0x0; // start address in slave
  telegram[0].u16CoilsNo = 1; // number of elements (coils or registers) to
read
 // telegram[0].u16reg = ModbusDATA; // pointer to a memory array in the
Arduino

and this section is to be added before calling
*xQueueReceive(modH->QueueTelegramHandle,
&telegram, portMAX_DELAY);*

I just want to confirm before I start modifying the code.

Regards

alejoseb, 2024-08-15:
I don't quite understand you are doing, notice that you don't need to modify the library. In the example that I pointed you can see that you send a telegram to the queue using a Modbus function.

nirbhow, 2024-08-16:
Thanks a lot for your response.
i think at the time of my previous email, i got little bit confused to
differentiate the purpose of two functions
*void StartTaskModbusMaster(void *argument) *in modbus.c
and
*void StartTaskMaster(void *argument) *in the freertos.c.
Now after you reply, what i have understood is that
*StartTaskMaster(void *argument) *is used to send telegram to queue
and the other function *StartTaskModbusMaster(void *argument) * is for
transfer of data from queue to USART.

am i right? or still missing something

Regards

alejoseb, 2024-08-21:
StartTaskMaster is the master handler task you don't need to worry about that unless you need to modify the library. So my recommendation is to let modbus.c as is. 

Please focus on the examples. Specifically, the following line shows how to send a master request using a specific telegram prepared earlier. Study the example first, and second the library if you need to mod it. 
https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/master/Examples/ModbusF103/Core/Src/freertos.c#L185

Closing this thread, since this is not an issue and I can only provide limited support.

closed 2024-08-21 by alejoseb

nirbhow, 2024-08-22:
Okay, thank you!