alejoseb/Modbus-STM32-HAL-FreeRTOS #112: modH->conn is NULL within TCPinitserver()

FebsterVF, 2024-09-18:
Hi, 

my implementation on an STM32H745-Disco dev board is stuck in the while loop in TCPinitServer, which is only entered if modH->conn is not initialized:

[image]

i tried replacing netconn_new() with the function it is referencing to: netconn_new_with_proto_and_callback(), but same problem.

here is the debug content of modH->conn:

[image]

i went through the code step by step, and it seems conn is not being initialized properly...

am i doing something wrong here?

alejoseb, 2024-09-19:
The only reason that I can imagine is if you are calling that function before the FreeRTOS kernel is initialized, as already mentioned in the code comment. I recommend you first have your STM32H745-Disco dev board running the TCP stack, after you confirm it is working then add the Modbus library. Besides, this is not an issue related to the library.

FebsterVF, 2024-09-23:
Thanks for the help, once i moved the MX_LWIP_Init(); to the section where all configured peripherals are initialized, i am no longer stuck.

It was in StartDefaultTask() by default...

Since the library is still not working for me, ill take a step back and look at just the TCP-Stack implementation as you recommended...

Thanks!

closed 2024-09-23 by FebsterVF