alejoseb/Modbus-STM32-HAL-FreeRTOS #115: Reconnect on Modbus TCP Master disconnect

FebsterVF, 2024-10-24:
Hi!

I was able to implement modbus tcp master and slave on an STM32H745-Discovery board, which works perfectly fine.

What im having trouble with is, once the master loses the connection to the slave, reconnecting does not work. At first the modbus thread got suspended indefinetly, until I removed the ulTaskNotifyTake call from the example right after the ModbusQuery call.

Now the connection can just be established on the startup of the µC, if the connection is interrupted, it is not established again, until the rest button is pressed.

I noticed there is a aging mechanism in the library file modbus.c and found that it depends on TCPAGINGCYCLES.

What i tried:

- Adjusted TCPAGINGCYCLES, tried values from 4 to 1000
- Adjusted osDelay in modbusTask, tried from 200ms to 1000ms
- changed TCP_TMR_INTERVAL in tcp_priv.h from 250 to 3, as described in another post here

what am i missing here? I need this to work reliable so it can properly handle connection loss...

FebsterVF, 2024-11-05:
I am still stuck with this...

does this work for anyone else, is this my setup or is this a known problem with the library?

alejoseb, 2024-11-06:
did you read this section https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS?tab=readme-ov-file#notes-and-known-issues-  

I have tested various scenarios disconnecting Ethernet cables for the slave and the master without any issue after changing manually the code as described in the readme

FebsterVF, 2024-11-06:
Thanks a lot, was able to make it work with these changes!

closed 2024-11-06 by FebsterVF