alejoseb/Modbus-STM32-HAL-FreeRTOS #118: problem in MODBUS SLAVE

nirbhow, 2024-12-18:
Dear Sir,
i'm trying to use your code libraries for modbus communication.
as master the code is working as per expectation.  
However, when i'm trying to use it for slave purpose i'm facing following issue:

The code stuck at 
line no. 680
at function   ulTaskNotifyTake(pdTRUE, portMAX_DELAY); /* Block until a Modbus Frame arrives */.
although the massage is transmitted from modbus master and it is received by modbus slave. The received message is available in ModbusH.xBufferRX.uxBuffer.

Regards

Luxiangxianghy, 2024-12-18:
> This is not an issue with the library please review the examples.

I had a similar problem.Referring to the example, I used my own development board as a slave and communicated with modbudpoll, and it was always timeout，The code is as follows：
  ModbusH.uModbusType = MB_SLAVE;
  ModbusH.port = &huart1;
  ModbusH.u8id = 3; //Modbus slave ID
  ModbusH.u16timeOut = 1000;
  ModbusH.EN_Port = NULL;
  ModbusH.u16regs = ModbusDATA;
  ModbusH.u16regsize= sizeof(ModbusDATA)/sizeof(ModbusDATA[0]);
  ModbusH.xTypeHW = USART_HW;
  //Initialize Modbus library
  ModbusInit(&ModbusH);
  //Start capturing traffic on serial Port
  ModbusStart(&ModbusH);

Luxiangxianghy, 2024-12-18:
> This is not an issue with the library please review the examples.

My usart is just a normal serial port, not connected to 485 or 232, how do I go about solving this problem?

nirbhow, 2024-12-18:
> This is not an issue with the library please review the examples.

Dear sir,
i'm not saying that there is issue with the library.
i'm  just asking what can be the reason for function 
ulTaskNotifyTake(pdTRUE, portMAX_DELAY); 
not getting unBlocked even when query/frame is received and is available in the ModbusH.xBufferRX.uxBuffer.
am i missing some setting. i have initialized the Modbus as 
ModbusH.uModbusType = MB_SLAVE;
  ModbusH.port =  &huart2; // This is the UART port connected to STLINK in the NUCLEO F429
  ModbusH.u8id = 1; //slave ID
  ModbusH.u16timeOut = 1000;
  ModbusH.EN_Port = NULL; // No RS485
  //ModbusH2.EN_Port = LD2_GPIO_Port; // RS485 Enable
  //ModbusH2.EN_Pin = LD2_Pin; // RS485 Enable
  ModbusH.u16regs = ModbusDATA;
  ModbusH.u16regsize= sizeof(ModbusDATA)/sizeof(ModbusDATA[0]);
  ModbusH.xTypeHW = USART_HW_DMA;

nirbhow, 2024-12-18:
which microcontroller are you using.
and at which particular point your code stuck.
i'm using STM32F730

alejoseb, 2024-12-18:
@nirbhow  line 680 will be unblocked once the DMA notification is executed, i.e., when a frame has arrived. This is according to your configuration, so I am assuming you are using the DMA mode (USART_HW_DMA).  In this case you need to verify that this notification https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/37f3fb5fa0ec1cfb3c45d3a5b5d683c177df669f/MODBUS-LIB/Src/UARTCallback.c#L152 is being executed. 

If you are not using the DMA mode (like @Luxiangxianghy's configuration) then you need to check that  this notification is being executed https://github.com/alejoseb/Modbus-STM32-HAL-FreeRTOS/blob/37f3fb5fa0ec1cfb3c45d3a5b5d683c177df669f/MODBUS-LIB/Src/Modbus.c#L46

If this notifications are not being executed, there could be multiple reasons, for example the DMA interrupts might not be enabled.
Also your FreeRTOS can be running out of heap memory.

Luxiangxianghy, 2024-12-19:
I've configured it like the example, but when I use modbus poll to communicate, I get a timeout!

alejoseb, 2024-12-19:
@Luxiangxianghy without more details it is unlikely that I can help you.
You need to check your whole setup and follow the instructions.

nirbhow, 2024-12-19:
Dear sir, 
its working now, actually i've forgot to include the uartcallback.c file in  the project.
hope that now i will be able to handle it.

regards

closed 2024-12-19 by nirbhow

Luxiangxianghy, 2024-12-20:
> @Luxiangxianghy without more details it is unlikely that I can help you. You need to check your whole setup and follow the instructions.

I followed the journey：  ModbusH.uModbusType = MB_SLAVE;
  ModbusH.port =  &huart1;
  ModbusH.u8id = 17; //Modbus slave ID
  ModbusH.u16timeOut = 1000;
  ModbusH.EN_Port = NULL;
  ModbusH.u16regs = ModbusDATA;
  ModbusH.u16regsize= sizeof(ModbusDATA)/sizeof(ModbusDATA[0]);
  ModbusH.xTypeHW = USART_HW;
  //Initialize Modbus library
  ModbusInit(&ModbusH);
  //Start capturing traffic on serial Port
  ModbusStart(&ModbusH);
  /* USER CODE END 2 */ But it's not responding， I use f103

Luxiangxianghy, 2024-12-20:
> @Luxiangxianghy without more details it is unlikely that I can help you. You need to check your whole setup and follow the instructions.

Does it need to be adjusted anywhere else?