alejoseb/Modbus-STM32-HAL-FreeRTOS #121: Looks interesting and useful

JimN555, 2025-02-25:
The work done here looks interesting and useful as I need a master and slave on one MCU.

That said I'm struggling to find any documentation.

Is any available?

How do you setup registers and coils in a slave?

Thanks

alejoseb, 2025-02-26:
This is a minimalist library. Thus, coils and registers share the same memory space in a given buffer that you need to provide while configuring the library. This means that you can access the same memory space as coils or registers. It is up to you to define specific memory ranges for registers or coils. 

Check the main.c file in any of the provided examples. The configuration section is there.  To see the actual usage of the library, check the freertos.c file. You can find examples for master and slave, and yes you can run both in the same MCU, as many as USARTs available in the MCU.

JimN555, 2025-02-26:
Hi Alejandro,

So far I do have the G431 example code working nicely. 

Next I will attempt to reconfigure it for a Slave.

If I can ask additional questions:
Do the ModbusQuery functions block other tasks from running?
If yes is there a way to avoid that and still be notified when the task is done? 

Thanks
Jim

alejoseb, 2025-02-26:
No it does not block other tasks, in fact the example uses notifications, it will yield while waiting for a response after executing a query.

closed 2025-02-28 by alejoseb