alejoseb/Modbus-STM32-HAL-FreeRTOS #122: Question: Why is USART needed?

NieLuc, 2025-02-28:
I want to use a STM32F765 with [ISOW1432 ](https://www.ti.com/lit/ds/symlink/isow1432.pdf?ts=1737449636729&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252Fde-de%252FISOW1432) to communicate with a RPi 3. 
Usually the RS485 protocol doesnt need a Clock Source and I haven't included one in my PCB design. However I do have connected the USART Pins to the ISOW1432. Do I need to worry about the Clocksource?

alejoseb, 2025-03-01:
Can you clarify your question? I don't know what you by "Clock source".

NieLuc, 2025-03-01:
Do you really need to use USART or can UART also be used?

alejoseb, 2025-03-01:
The library currently uses USART in the asynchronous mode (i.e., without source clock). Thus, it is compatible with UART, but I have not tested myself.

NieLuc, 2025-03-01:
Okay, thank you!
Is the Drive Enable Pin/Hardware Flow Control Pin needed for RS485 or can a random Output be used as a Drive enable pin?

alejoseb, 2025-03-01:
Any pin can be used.

NieLuc, 2025-03-01:
Oh okay, thank you again! :)
Is there a example for this with a driver enable pin?

closed 2025-03-01 by alejoseb