alejoseb/Modbus-STM32-HAL-FreeRTOS #125: Slave stops responding after months of operation (ID5 issue)

cuong19654, 2025-06-02:
Hi, thank you very much for sharing this Modbus RTU project.
I have been using it successfully in my system with 5 slave devices (ID1 to ID5), all running the same firmware.

For the first 3 months, everything worked perfectly. However, recently I found that:

ID1 to ID4 still work fine.

ID5 stops responding even though the master still sends valid frames.

I'm using UART with interrupt mode (not DMA), and the frame sent to ID5 is valid with correct CRC.

Could you please let me know:

Have you ever experienced this issue where one slave (ID5) suddenly stops responding?

Do you have any suggestions on what might be causing the task to hang or miss frames?

I really appreciate your work and would be grateful for any advice.

alejoseb, 2025-06-03:
Hi, I have never experienced that issue. There could be several reasons for this to happen, which are unrelated to this library. 

I will name a few: 
- the modbus task run-outs of heap memory, you can adjust the RTOS configuration to solve that
- Electrical damage. If after rebooting the slave the error persists, you need to replace the slave.
- a bug in another task

Hope you solve the issue, good luck 🤞

closed 2025-06-03 by alejoseb

cuong19654, 2025-06-10:
Hi, thank you very much for your suggestions.
I’d like to share what I’ve confirmed so far:

I checked the stack usage of the Modbus slave task using uxTaskGetStackHighWaterMark(), and found that only 17 words (i.e., 68 bytes) remain.
When I intentionally caused a stack overflow in that task, the entire system became unstable, and other tasks also behaved abnormally.
However, in the actual issue we're facing, only the device with ID5 stops responding to Modbus commands, while other devices (ID1 to ID4) continue to operate normally.
Furthermore, even after Modbus communication stops, the device with ID5 continues to operate independently — its main application is still running fine.
This suggests the issue is limited to the Modbus communication part, not a system-wide failure.

As for the heap, I believe that if it were exhausted, it would affect the entire system, not just one slave device.

Therefore, I suspect the Modbus task or UART receive handling in the device with ID5 silently stopped processing incoming frames, possibly due to a UART reception error or an internal state not resetting properly.

If you have any further ideas, I’d be happy to try them.
Thank you again for your excellent work and kind support.